ros-drivers/joystick_drivers #72: Jade Release

wjwwood, 2015-01-15:
I wanted to go ahead and start the Jade discussion here, since we are probably not going to import `qtsixa` for Utopic and Vivid, see: https://github.com/ros/rosdistro/issues/6435#issuecomment-70165912

Is this going to be an issue or has this package moved on from using `qtsixa`? The only reason it is on our radar is that we were importing it for other Ubuntu distros in the past.

mikeferguson, 2015-04-12:
Nothing in this repo requires qtsixa -- qtsixa was basically an alternative to ps3joy.

vrabaud, 2015-05-24:
so we're all good then ? @jbohren , should we release for Jade ? I can if you want, using the indigo-devel banch.

mikeferguson, 2015-05-24:
@vrabaud I actually released this into indigo, so I'll go ahead and push a jade release in a few minutes.

vrabaud, 2015-05-24:
thx ! I saw there were commits after 1.10, not sure they are critical though: https://github.com/ros-drivers/joystick_drivers/commits/indigo-devel

mikeferguson, 2015-05-24:
yeah -- I'm tagging a new release right now -- mainly for the fedora guys

mikeferguson, 2015-05-24:
https://github.com/ros/rosdistro/pull/8542

closed 2015-05-24 by mikeferguson