ros-drivers/joystick_drivers #74: ps3joy.py problem to activate connection

hannesharms, 2015-03-16:
If I run ps3joy.py on ros indigo and pressing the pairing button nothing happens. (no output, no error, no connection activated).  
If I remove the bluetooth dongle and plug in agian, while ps3joy.py still waiting, the connection is activated and node runs normally.
I tested this behavier with two diffrent bluetooth adapters.

hannesharms, 2015-03-16:
Seems to be a bug in bluez 4.1 . Upgrading to bluez  5 via ppa solves to problem.

jprod123, 2017-06-20:
This is a known bug. I restart bluetooth via sudo systemctl restart bluetooth while ps3joy is still waiting and it works. For some reason bluez crashes when launching the script. On bluez 5.37 by the way

Tengyun-Mo, 2019-07-22:
I can't found ps3 via device search.I don't where the problem is?Who can help me?I'm very appreciate it.