ros-drivers/joystick_drivers #76: spacenav: tutorials outdated

gavanderhoorn, 2015-07-25:
The tutorials linked from the [spacenav_node](http://wiki.ros.org/spacenav_node) wiki page still refer to rosbuild and seem to suggest that the package should be build from source. I've seen two questions about this on ROS Answers in a few weeks:

 - [Tutorials / Using the SpaceNav Node doesn't work](http://answers.ros.org/question/214537/tutorials-using-the-spacenav-node-doesnt-work/)
 - [3d mouse with ros](http://answers.ros.org/question/213428/3d-mouse-with-ros/)

John-Nagle, 2015-07-25:
gvdhoom left a good answer at the wiki page, which could be used to revise the tutorials.
Also worth noting is that the Ubuntu install seems to install the driver daemon as a daemon, so the hardware is recognized without having to run a driver program each time.

These devices can be modified to act as 6-DOF force sensing wrists. So please consider supporting more than one instance. Low priority.

Worth noting: /spacenav/offset contains the 3 translations, range -0.75 to 0.75, and /spacenav/rot_offset contains the 3 rotations, same range. /spacenav/joy contains the three translations followed by the 3 rotations. The first two can be graphed with rqt_plot, but "joy" cannot.
[image]
Image description: rqt_plot showing 3DConnexion mouse output. Mouse turned clockwise to limit.

JWhitleyAStuff, 2019-12-17:
The whole stack is outdated. Closing.

gavanderhoorn, 2019-12-18:
@JWhitleyAStuff: how is this a proper resolution of the issue?

JWhitleyAStuff, 2019-12-18:
@gavanderhoorn Sorry, this was more of a bad joke. Reversing.