ros-drivers/joystick_drivers #77: joy_node frequency/rate parameters

jvgomez, 2015-08-17:
Hi there,

I think that the parameters _autorepeat_rate_ and _coalesce_interval_ should be both given in the same units (first is in Hz and second in ms). My personal preference would be both to be a rate. This would require obviously to change the name of the parameter to _coalesce_rate_

The reason is that, apart from consistency, I think both parameters are related. I (and probably everybody) use the _coalesce_interval_ to set a maximum rate for the joy message. And then _autorepeat_rate_ keeps that rate when the input rate is lower. Now, if I want to have a constant rate for the joy message, I have to set 2 parameters in my launch files, which would be OK if they where both same units, so I do not have to have two different parameters with different values.

Therefore, I propose this as an enhancement.

JWhitleyAStuff, 2018-12-07:
Are you OK with this going in for Lunar and above? Breaking the API for older versions seems unnecessary for a semantic bug.

jvgomez, 2018-12-07:
Sure. This was 3 years ago, I do not even work with ROS anymore hehe

closed 2019-12-17 by JWhitleyAStuff