ros-drivers/joystick_drivers #84: Dualshock 4 support

ibaranov-cp, 2016-01-14:
Hello all,

Does this package support, or plan to support soon, the Dualshock 4?

Dualshock 3 is EOL.

tonybaltovski, 2016-03-28:
I just wanted to follow up with this.  I have confirmed that the [ds4drv](https://github.com/chrippa/ds4drv) works with the PS4 controller when in `hidraw` mode as a regular input.  I spun up a VM and using Atheros AR3011 USB adaptor. It can be installed via pip using `sudo pip install ds4drv`.  Pairing can be done by enabling pairing on controller by holding Share + the PS button until the LED starts blinking rapidly.  Scan for a controller using `hcitool scan` and a "Wireless controller xx.xx.xx.xx" should be listed.  Pair with `sudo bluez-simple-agent hci0 xx.xx.xx.xx` and trust the device with `sudo bluez-test-device trusted xx.xx.xx.xx yes`  or use a ([pairing script](https://gist.github.com/tonybaltovski/011e2be2e5d41db6dbec)).  A daemon service can be added using [this](https://gist.github.com/tonybaltovski/a6f0cb11ddad73e46a26) with this [config](https://gist.github.com/tonybaltovski/fc4fbfdb4d715e2dcec8).

Any thoughts on going forward deploying the PS4 controller into the ROS ecosystem?  Other bluetooth controllers suggestions welcomed.

rbbg, 2017-02-24:
I also wanted to use the DS4 with ROS and found it surprisingly easy. After installing ds4drv and pairing the controller, I ran `ds4drv --hidraw`, which mounted the controller onto `/dev/input/js1`. Then, running a roscore and `rosparam set joy_node/dev "/dev/input/js1"` + `rosrun joy joy_node` spun up a functional node publishing all the axes and button states.

The mapping of the buttons/axes was a bit strange, but it's definitely usable.

davefeilseifer, 2019-06-12:
added files for ps4 controller mappings when using ds4drv to pr2_teleop (https://github.com/PR2/pr2_apps/pull/37)

JWhitleyAStuff, 2019-06-12:
Closing issue thanks to resolution by @rbbg.

closed 2019-06-12 by JWhitleyAStuff

naoki-mizuno, 2019-09-01:
I want to add that I wrote [`ds4_ros`](http://wiki.ros.org/ds4_ros), a ROS driver for DualShock 4 that allows you to access IMU and touchpad info and also send feedback to the device.