ros-drivers/joystick_drivers #90: Separate autorepeat_rate for buttons and axes

hvpandya, 2016-09-15:
In most joystick applications I have come across, autorepeat_rate comes in handy while using axes for operating robot at constant velocity. 

But if I want to use some of the buttons to toggle modes, autorepeat_rate kind of kills the purpose.

Having separate params for both can provide more flexibility to end user for mapping axes/buttons for various cases.

JWhitleyAStuff, 2018-06-11:
Since the `sensor_msgs/Joy` message includes both a buttons array and an axes array, you cannot publish one without the other (unless you created a new message type). What would you suggest be published in the buttons array? An empty array?

JWhitleyAStuff, 2019-01-22:
Closing due to lack of activity.

closed 2019-01-22 by JWhitleyAStuff