ros-drivers/joystick_drivers #91: Wrong reading of axes util their value changes

omcandido, 2016-10-18:
I am using a steering wheel to control a car in V-REP and I have observed that until I push the brake or the accelerator, I get a value of zero by default. Both pedals, without being pressed should have a value of -1. This makes the car automatically brake or speed up.

It would be nice if the right values were correctly read from start.

1r0b1n0, 2016-12-01:
I have the same issue with joy_node, using the "Microsoft Xbox 360 Wired Controller"
Removing this line fixes the problem for me : (joy_node.cpp : 244)
`if (!(event.type & JS_EVENT_INIT)) // Init event.value is wrong.`

kmhallen, 2018-06-26:
@omcandido @1r0b1n0 Has this been fixed with the latest binary? Can we close this issue?

1r0b1n0, 2018-06-26:
@kmhallen I tested with the Xbox 360 wired controller;  setting the "default_trig_val" parameter to true fixes the issue.
Thanks !

kmhallen, 2018-06-26:
@JWhitleyAStuff Can you close this issue?

closed 2018-07-03 by JWhitleyAStuff