ros-drivers/joystick_drivers #92: When the connection with the wireless receiver is lost, the last received command is sent infinitely

ItsJas, 2016-11-28:
When the joystick is out of  range and unable to communicate with the wireless receiver, the last received command is sent infinitely!
This is really dangerous especially when working with mobile robots, there should be a way to indicate that the joystick is out of range. If so, a "stop" command with no movements should be sent and not the last received command.There should be a warning and a state error indicator so that the situation could be handled.

peci1, 2017-03-08:
What would you consider a "stop" message in terms of `sensor_msgs/Joy`? Publishing all zeros is no option, since on e.g. LB/LR buttons, zero means half-pressed.

jacobperron, 2017-04-19:
At the very least any button presses reported can be zero. This way a safety can be implemented via a deadman button, like in [joy_teleop](https://github.com/ros-teleop/teleop_tools) (e.g. only send commands if X is pressed).

pirobot, 2018-06-05:
This happened to our very expensive outdoor mobile robot a few days ago and it drove itself off a curb.  Fortunately the robot was undamaged but it could have been much worse.  I need to fix this ASAP so could someone please at least point me to where in joy_node.cpp I might be able to insert a check to see if the connection has been lost? Thanks!

peci1, 2018-06-05:
@pirobot Maybe periodic checks if the joy device (`/dev/js0` and so on) is present?

peci1, 2018-12-25:
If autorepeat is not used, then the solution is easy - just stop publishing (and hope the receiving end understands that signals a problem). 

In autorepeat case, I can imagine two approaches:

1. Add new parameters where the user would set the "disconnect" value of all the buttons and axes. And the default could be all zeros. That'd work in most cases (but wouldn't help e.g. in our case when we have multiple types of joysticks and you never know which one (and in which mode) is to be connected).
1. Make the joy_node adaptive. It'd store the first message it sends after being started or after a joystick is (re)connected, and send that message when the joystick is lost. This would have the benefit of being configuration-less, but would require that when launching the node, the joystick will be in a state when no buttons are physically pressed. For me, that'd be acceptable. But I'm not sure if it'd be so for everyone.

NickL77, 2018-12-26:
I did a little bit of testing and came to some findings:

1. If the joy_node is running on a separate machine, and the connection between the machines are lost, there isn't much that we can do. The subscriber needs to find a way to check if a message has not been delivered. (Best from what I can think of is to use autorepeat and verify that a message is being published at the autorepeat rate)

2. Correct me if I'm wrong, but I'm assuming that most people are encountering an issue when the joystick itself is disconnected from joy_node, resulting in the following error message `Connection to joystick device lost unexpectedly. Will reopen.` When this is the case, even if autorepeat is on, the node will not continue looping and publishing new messages, so we can simply send out the "first message" once.

If I am correct, the solution to this issue/enhancement is to simply send out the "first message" after the joystick is disconnected from the node. All other cases should be handled by using autorepeat and having the subscribers expect new messages to be published at the given rate.

pirobot, 2019-01-07:
The best way I know of to simulate our problem case is to set the robot in motion by holding an input on the gamepad (better done in Gazebo for safety), then remove one of the batteries from the gamepad.  You will see that the robot keeps moving for several seconds and the last input continues to be published on /cmd_vel (or whatever topic you are using) during the time the robot keeps moving.  The status of /dev/input/js0 during this time remains unchanged.

NickL77, 2019-01-10:
@pirobot So if I'm understanding your problem correctly:
1. You are running the joystick node with autorepeat on
2. When you remove the batteries from your controller, there is no error, and `/dev/input/js0` is still active
3. The node continues to publish the last message for several seconds until there is an error and `/dev/input/js0` is no longer connected

pirobot, 2019-01-11:
@NickL77 Almost but not quite.  For number 3 above, there is no error and the device /dev/input/js0 does not disappear.  In fact, I can have jstest running and the last inputs continue to display until about 3 seconds, then they all go to zero.

Also, on the Logitech F710 that I have been using, I have been setting the little slide switch set to "X".  If I set it to "D", then the first time I pull the batteries, I get the desired behavior; namely, all inputs go to zero as soon as I remove a battery.  However, if put all batteries back in and try removing one again, it goes back to the delayed behavior.  The same thing happens if I then set the slide switch back to "X"; namely, the first time I remove a battery, all inputs go to zero but if I put them back in and then remove a battery, the reaction is delayed.

peci1, 2019-01-22:
The problem seems to be difficult because the behavior of the drivers of various gamepads is inconsistent.

I tested with these 3 (4) wireless gamepads:

* Logitech Rumblepad 2
* Logitech F710 XInput mode
* Logitech F710 DirectX mode
* Xbox One bluetooth controller

I recorded bagfiles with these topics of interest:

* /joy
* /rosout (to see error messages)
* /usb_cam/image_raw/compressed (live video of me fiddling with the gamepad)
* /screencast/compressed (screencast of a terminal session with various useful printouts; this one is not very precisely synchronized with the rest (give it +- 1 sec))

The bag files are uploaded to [Google drive](https://drive.google.com/drive/folders/1GjvEjP2zfxhq_mLft2aGuHcBEp0LOnSO) (cca 2 GB because the screencast is converted to the compressed transport). There are two subfolders: autorepeat and no_autorepeat to see how the behavior changes when using autorepeat_rate.

I'll try to summarize my findings in a later post, but generally, the behavior is inconsistent.

peci1, 2019-01-23:
Here's the resume:

For each of the controllers, I tested both removing the battery and unplugging the dongle from USB (or turning off Bluetooth in case of Xbox One).

The tested categories are:

- zero after unplug: whether all zeros are published when the battery/dongle gets unplugged
- correct after replug: whether the driver automatically provides correct values after replug even in case the axis has moved while unplugged (`x` means the old value is still there, `zeros` mean it publishes zeros until first move, `ok` means the correct value is published right after reconnecting)
- axes move after replug: whether axis motion is detected after replugging the controller; `after button press` means you first have to press a button and only after that does the driver register axis motion
- sending data: (only for autorepeat) whether joy messages are being periodically published when the battery/dongle is unplugged

## autorepeat

| | RumblePad2 | F710 D | F710 X | Xbox One |
| --- | --- | --- | --- | --- |
| zero after battery unplug | x | 1s | 1s | x |
| zero after dongle unplug | x | x | x | x |
| correct after battery replug | x | zeros | ok | N/A |
| correct after dongle replug | ok | zeros | zeros | zeros |
| axes move after battery replug | after button press | ok | ok | N/A |
| axes move after dongle replug | ok | ok | ok | ok |
| sending data: battery unplugged | ok | ok | ok | ok (until BT is lost; 20s) |
| sending data: dongle unplugged | x | x | x | x (BT off) |

## no autorepeat

| | RumblePad2 | F710 D | F710 X | Xbox One |
| --- | --- | --- | --- | --- |
| zero after battery unplug | x | 1s | 1s | x |
| zero after dongle unplug | x | x | x | x |
| correct after battery replug | x | ok | ok | N/A |
| correct after dongle replug | x | zeros | zeros | zeros |
| axes move after battery replug | after button press | ok | ok | N/A	 |
| axes move after dongle replug | after button press | ok | after button press | ok |

Concluding, removal of the dongle always results in autorepeat messages not being published at all. After battery unplug, some controllers will publish 0 after a second, others will not (and there's really no way of being notified of this event because there's nothing at all in `cat /dev/input/js0` after removing the battery).

Furthermore, some controllers have quirks regarding reconnection and the values they produce afterwards, but if we'll assume the users will either reconnect the controllers in their "zero state" or they'll fiddle a bit with the axes/buttons after reconnection, this can be neglected.

So, e.g. for RumblePad 2, there's plainly no way to tell when it got disconnected. And thus any kind of reaction to that event is impossible.

JWhitleyAStuff, 2019-11-01:
Thanks a ton to @peci1 for all of the testing! Closing this as `wontfix`.

closed 2019-11-01 by JWhitleyAStuff