ros-drivers/joystick_drivers #93: Connection Feedback

shoemakerlevy9, 2017-02-03:
Minor Issue, but I've noticed that when using on a Jetson TX1 (Ubuntu 14.04 64-bit embedded) the 4 central LED's always flash. But everything seems to work fine. 

With my laptop (Dell XPS 15 Ubuntu 14.04 64-bit) the controller indicates that its player 1.

It would be nice to have this feedback implemented for a TX1 since they are so nice to build robots with. I'll look into a fix but I'm new fairly new to using this driver.

JWhitleyAStuff, 2018-12-07:
@shoemakerlevy9 - Sorry it's been a long time since this was attended to. I have seen other issues with the Jetson boards and button mapping. Is this something you have seen as well?

JWhitleyAStuff, 2019-07-18:
Closing very old issue.

closed 2019-07-18 by JWhitleyAStuff