ros-drivers/joystick_drivers #97: Compile from scratch in Kinetic

gyang1011, 2017-04-24:
I am trying to compile this package from scratch in my catkin_ws/src using catkin_make_isolated and I keep getting the following error:

CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/burobotics/anaconda2/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/burobotics/catkin_ws/build_isolated/joy/catkin/catkin_generated/version/package.cmake")
  returned error code 1

I am using ros kinetic and ubuntu 16.04

JWhitleyAStuff, 2018-06-11:
Have you properly sourced your environment? Try pulling again (several updates have been made), cleaning your workspace (catkin_make clean), and try building again.

JWhitleyAStuff, 2018-07-03:
Closing after a month with no activity.

closed 2018-07-03 by JWhitleyAStuff