ros-drivers/joystick_drivers #99: Lunar release please?

bmagyar, 2017-05-01:
Hi @mdhorn,

Could we get a Lunar release please? Is there something blocking it?

BrunoXL, 2017-06-04:
According to ros status page (http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=joy&s=6&r=1) alll dependencies have been releases.
I'm also interested in a Lunar release.

mikaelarguedas, 2017-06-17:
:+1: that would be great to have this package available in Lunar

mikaelarguedas, 2017-07-11:
@jbohren Any update on this ? (I see that all PRs have been merged :smiley:)

mikaelarguedas, 2017-08-05:
@jprod123 Is there anything we can do to help getting this released? this is currently the repository [blocking the most packages](http://repositories.ros.org/status_page/blocked_releases_lunar.html?s=6&r=1) from being released into lunar

mikaelarguedas, 2017-08-21:
@jbohren @jprod123 FYI I'm planning on releasing the current kinetic version (1.11.0-0) into Lunar tomorrow.
Please comment here if there is a reason not to do so.

brleme, 2017-08-23:
@mikaelarguedas , did you work on it?

mikaelarguedas, 2017-08-23:
@brleme Yes, I added the missing rosdep keys. One PR is [pending](https://github.com/ros/rosdistro/pull/15808), once it's merged I'll make the release

mikaelarguedas, 2017-08-23:
released in https://github.com/ros/rosdistro/pull/15813

closed 2017-08-23 by mikaelarguedas

cdancette, 2017-08-24:
Is the package including joy is in the ubuntu repos already ? 
How can we use joy in the meantime if it's not ?

mikaelarguedas, 2017-08-24:
The package is in the testing repository for now and will be available in the main repository at the next sync (we try to sync every 2-3 weeks). A good way to track the status of a ROS package is to look at the ROS status pages (e.g. http://repositories.ros.org/status_page/ros_lunar_default.html?q=joy).

To be able to use it there are a few options:
-  [add the testing repository to your sources](http://wiki.ros.org/ShadowRepository) and you will be able to run `sudo apt update && sudo apt install ros-lunar-joy`. 
- wait for the next sync (announced on discourse.ros.org, [example here](https://discourse.ros.org/t/new-packages-for-lunar-2017-08-22/2490)).
- clone this repository and build the packages from source

Hope this helps