ros-drivers/joystick_drivers #114: ps3joy overlaps my mouse behavior

Maik93, 2017-10-30:
Hi,
I've recently installed this package and I'm facing this problem: once my PS3 joystick is connected (through the node `ps3joy_node.py` or `ps3joy.py`) my mouse stops to listen any click (I can't close or select windows or anything) and joystick's analog left acts like mouse' scroller.
The topic `/joy` however publish correctly what the joystick does, so I can control my robot i.e. in Gazebo, but while controlling it, the camera moves (like it's been dragged by the mouse).
Anyone experiencing this issue?  Sorry if it's a well known argument, but i haven't found anything in this issue-tracker.

Using Ubuntu 16.04 LTS and ROS Lunar, original PS3 controller.

0xd1ma, 2018-01-30:
I have a similar problem with DualShock2 after upgrading Ubuntu.

JWhitleyAStuff, 2018-08-24:
I don't have a timeline but I'll see if I can investigate this. Are you using Bluetooth or USB for the connection?

Maik93, 2018-08-24:
Tried both, but mainly I use USB

pranjalipadghan, 2019-01-18:
Same problem.

fgrigis, 2019-01-25:
same problem

crn-wpi, 2019-11-04:
same problem

mmurooka, 2019-11-14:
same problem

svdeepak99, 2019-12-21:
Hi guys. I have a solution to this. You can disable the joystick input device, using `xinput` command in the terminal and then the joystick will not overlap with the mouse.

I followed the following steps:
`xinput -list`
With this I found that my PS3 joystick's id was 25. And hence I disabled the device using the following command:
`xinput --disable 25`
You can replace the number 25, with your corresponding device id found using `xinput -list` command.

Now the joystick works perfectly for me, without overlapping with the mouse.

umhan35, 2021-02-04:
@svdeepak99 Thanks! It worked. The ID for me is 15, which is under "Virtual core pointer".

closed 2021-02-04 by JWhitleyWork