ros-drivers/joystick_drivers #118: New Release for Kinetic

tdenewiler, 2018-05-01:
I was wondering if there are plans to make a new release for Kinetic? There are a fair amount of commits that have been merged since v1.11.0. The one that most affects me is #98 to initialize the joystick commands. It would be nice to get that fix from apt. Thanks.

JWhitleyAStuff, 2018-06-11:
New release 1.12.0 is pending approval and package sync for Indigo, Kinetic, Lunar, and Melodic. Should be in ros-shadow-fixed in a couple of hours (with the exception of Melodic which is currently mid-sync). I'll update this issue once they've been sync'd to the primary repos.

kmhallen, 2018-06-26:
The sync is complete: https://discourse.ros.org/t/new-packages-for-kinetic-2018-06-22/5173
This can probably be closed

closed 2018-06-26 by JWhitleyAStuff