ros-drivers/joystick_drivers #119: Release for Melodic

kmhallen, 2018-05-02:
It looks like all of the dependencies are available in Melodic, so it would be great to get this released. Thanks in advance!

gvallicrosa, 2018-06-08:
I just tried it in a virtual machine with ubuntu 18.04 and ROS melodic and compiled fine.

JWhitleyAStuff, 2018-06-11:
Release 1.12.0 is pending for Indigo, Kinetic, Lunar, and Melodic. See #118 for updates.

closed 2018-06-11 by JWhitleyAStuff

roehling, 2018-06-26:
The release 1.12.0 has been published for all ROS versions except Melodic, which is the only ROS version with no released `joystick_drivers` whatsoever. Can someone with more insight than me figure out why the Melodic release is missing?

kmhallen, 2018-06-26:
It has been released into Melodic and it is present in the  ```building``` and ```testing``` repos. It will make it to ```main``` when the person at OSRF decides to sync all package to ```main```.
http://repositories.ros.org/status_page/ros_melodic_default.html?q=joy

The link below explains how to pull binaries from repos other than main/public:
http://wiki.ros.org/ShadowRepository

roehling, 2018-06-27:
Thanks, I am aware of the shadow repository; however, I'm planning to migrate several departments to the new ROS LTS, and I'd rather have them not tinker around with release candidates. 

I was just wondering if there might be a particular reason for the missing release besides "we didn't feel like syncing it, man". :wink: