ros-drivers/joystick_drivers #129: Segmentation Fault

AldrichCabrera, 2018-09-14:
Hello, I have a question. When I launch Rosrun joy joy_node immediately display the error: **Segmentation Fault.** I've installed all dependencies and driver but don't work. The joystick is a PS3 and this joystick is connected by Bluetooth and work, the dev is /js0. Just the Ros node doesn't work.

strongly-typed, 2018-09-25:
I can somehow reproduce this in a Docker container on three different machines:

```
docker run --interactive --tty --rm --cap-add=SYS_PTRACE --security-opt seccomp=unconfined  ros:kinetic-ros-base /bin/bash
```

Inside of the container:
```
apt update && apt install ros-kinetic-joy gdb strace -y
roscore&
gdb /opt/ros/kinetic/lib/joy/joy_node
```

Let it run and get a `SIGSEGV` error:
```
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/kinetic/lib/joy/joy_node...(no debugging symbols found)...done.
(gdb) run
Starting program: /opt/ros/kinetic/lib/joy/joy_node
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1855700 (LWP 657)]
[New Thread 0x7ffff1054700 (LWP 658)]
[New Thread 0x7ffff0853700 (LWP 659)]
[New Thread 0x7fffebfff700 (LWP 664)]

Thread 1 "joy_node" received signal SIGSEGV, Segmentation fault.
0x00007ffff73f3dda in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
(gdb) backtrace
#0  0x00007ffff73f3dda in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
#1  0x0000000000415b4b in Joystick::Joystick() ()
#2  0x000000000040e733 in main ()
```

On the host (Ubuntu 16.04.5 LTS), this works, even with no joystick connected.
```
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/kinetic/lib/joy/joy_node...(no debugging symbols found)...done.
(gdb) run
Starting program: /opt/ros/kinetic/lib/joy/joy_node
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1856700 (LWP 7591)]
[New Thread 0x7ffff1055700 (LWP 7592)]
[New Thread 0x7ffff0854700 (LWP 7593)]
[New Thread 0x7fffebfff700 (LWP 7598)]
[ERROR] [1537881293.102796193]: Couldn't open joystick /dev/input/js0. Will retry every second.
```

I added `--cap-add=SYS_PTRACE` and `--security-opt seccomp=unconfined` for purposes of debugging. It also crashes without these options.

gurbain, 2018-09-28:
Hi,

I have the same problem here, since I updated the joy node package:
`$ apt-cache show ros-kinetic-joy
Package: ros-kinetic-joy
Version: 1.12.0-0xenial-20180824-113827-0800
Architecture: amd64
Maintainer: Jonathan Bohren <jbo@jhu.edu>
Installed-Size: 163
Depends: libboost-system1.58.0, libc6 (>= 2.15), libgcc1 (>= 1:3.0), libstdc++6 (>= 5.2), joystick, ros-kinetic-diagnostic-updater, ros-kinetic-roscpp, ros-kinetic-sensor-msgs
Homepage: http://www.ros.org/wiki/joy
Priority: extra
Section: misc
Filename: pool/main/r/ros-kinetic-joy/ros-kinetic-joy_1.12.0-0xenial-20180824-113827-0800_amd64.deb
Size: 45820
SHA256: bfda09fe0170a55d24eab58f2d275539b53225f8895f40ae11f53795d47c7d46
SHA1: 30bbe6171214cfbc1b4b960849b868b698667e42
MD5sum: eea584c69f5c0a58eb03b4de9b687494
Description: ROS driver for a generic Linux joystick.
 The joy package contains joy_node, a node that interfaces a generic Linux joystick to ROS. This node publishes a &quot;Joy&quot; message, which contains the current state of each one of the joystick's buttons and axes.
Description-md5: e2db4e8acf0aba1c676ef190b397f7dd`

Ubuntu version:
`$ lsb_release -a
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.4 LTS
Release:	16.04
Codename:	xenial
`

Debugging output:
`(gdb) run
Starting program: /opt/ros/kinetic/lib/joy/joy_node 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1a67700 (LWP 27486)]
[New Thread 0x7ffff1266700 (LWP 27487)]
[New Thread 0x7ffff0a65700 (LWP 27488)]
[New Thread 0x7fffebfff700 (LWP 27493)]

Thread 1 "joy_node" received signal SIGSEGV, Segmentation fault.
0x00007ffff73f3dda in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
(gdb) 
`

strongly-typed, 2018-09-30:
TLDR: Update everything, seems to be gone now.

It is still reproducible using the docker container
```
ros                       kinetic-ros-base    d7931dc419c7        3 weeks ago         1.19GB
```
This container is still the latest:
```
$ docker pull ros:kinetic-ros-base
kinetic-ros-base: Pulling from library/ros
Digest: sha256:75979f4cd3665fafe0b5af1acccdce395fa87ceeb631697b855c06b1dfc9c693
Status: Image is up to date for ros:kinetic-ros-base
```

Although, when I update all ROS packages inside of the container, the error is gone:

```
$ docker run --interactive --tty --rm --cap-add=SYS_PTRACE --security-opt seccomp=unconfined  ros:kinetic-ros-base /bin/bash
root@cb77897108d7:/# apt-get update && apt-get dist-upgrade -y -qq && apt-get install ros-kinetic-joy gdb
Get:1 http://security.ubuntu.com/ubuntu xenial-security InRelease [107 kB]
Get:2 http://archive.ubuntu.com/ubuntu xenial InRelease [247 kB]
Get:3 http://packages.ros.org/ros/ubuntu xenial InRelease [4040 B]
Get:4 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [109 kB]
Get:5 http://security.ubuntu.com/ubuntu xenial-security/universe Sources [93.8 kB]
Get:6 http://archive.ubuntu.com/ubuntu xenial-backports InRelease [107 kB]
Get:7 http://security.ubuntu.com/ubuntu xenial-security/main amd64 Packages [709 kB]
Get:8 http://archive.ubuntu.com/ubuntu xenial/universe Sources [9802 kB]
Get:9 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages [688 kB]
Get:10 http://security.ubuntu.com/ubuntu xenial-security/restricted amd64 Packages [12.7 kB]
Get:11 http://security.ubuntu.com/ubuntu xenial-security/universe amd64 Packages [481 kB]
Get:12 http://security.ubuntu.com/ubuntu xenial-security/multiverse amd64 Packages [3744 B]
Get:13 http://archive.ubuntu.com/ubuntu xenial/main amd64 Packages [1558 kB]
Get:14 http://archive.ubuntu.com/ubuntu xenial/restricted amd64 Packages [14.1 kB]
Get:15 http://archive.ubuntu.com/ubuntu xenial/universe amd64 Packages [9827 kB]
Get:16 http://archive.ubuntu.com/ubuntu xenial/multiverse amd64 Packages [176 kB]
Get:17 http://archive.ubuntu.com/ubuntu xenial-updates/universe Sources [280 kB]
Get:18 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages [1097 kB]
Get:19 http://archive.ubuntu.com/ubuntu xenial-updates/restricted amd64 Packages [13.1 kB]
Get:20 http://archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages [889 kB]
Get:21 http://archive.ubuntu.com/ubuntu xenial-updates/multiverse amd64 Packages [18.8 kB]
Get:22 http://archive.ubuntu.com/ubuntu xenial-backports/main amd64 Packages [7343 B]
Get:23 http://archive.ubuntu.com/ubuntu xenial-backports/universe amd64 Packages [8256 B]
Fetched 26.3 MB in 3s (8482 kB/s)
Reading package lists... Done
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 59739 files and directories currently installed.)
Preparing to unpack .../libglib2.0-0_2.48.2-0ubuntu4.1_amd64.deb ...
Unpacking libglib2.0-0:amd64 (2.48.2-0ubuntu4.1) over (2.48.2-0ubuntu4) ...
Preparing to unpack .../tzdata_2018e-0ubuntu0.16.04_all.deb ...
Unpacking tzdata (2018e-0ubuntu0.16.04) over (2017c-0ubuntu0.16.04) ...
Preparing to unpack .../libcurl3-gnutls_7.47.0-1ubuntu2.9_amd64.deb ...
Unpacking libcurl3-gnutls:amd64 (7.47.0-1ubuntu2.9) over (7.47.0-1ubuntu2.8) ...
Preparing to unpack .../libglib2.0-data_2.48.2-0ubuntu4.1_all.deb ...
Unpacking libglib2.0-data (2.48.2-0ubuntu4.1) over (2.48.2-0ubuntu4) ...
Preparing to unpack .../binutils_2.26.1-1ubuntu1~16.04.7_amd64.deb ...
Unpacking binutils (2.26.1-1ubuntu1~16.04.7) over (2.26.1-1ubuntu1~16.04.6) ...
Preparing to unpack .../libcurl3_7.47.0-1ubuntu2.9_amd64.deb ...
Unpacking libcurl3:amd64 (7.47.0-1ubuntu2.9) over (7.47.0-1ubuntu2.8) ...
Preparing to unpack .../liblcms2-2_2.6-3ubuntu2.1_amd64.deb ...
Unpacking liblcms2-2:amd64 (2.6-3ubuntu2.1) over (2.6-3ubuntu2) ...
Preparing to unpack .../linux-libc-dev_4.4.0-135.161_amd64.deb ...
Unpacking linux-libc-dev:amd64 (4.4.0-135.161) over (4.4.0-134.160) ...
Preparing to unpack .../ros-kinetic-genmsg_0.5.11-0xenial-20180824-080325-0800_amd64.deb ...
Unpacking ros-kinetic-genmsg (0.5.11-0xenial-20180824-080325-0800) over (0.5.10-0xenial-20180809-135325-0800) ...
Preparing to unpack .../ros-kinetic-gencpp_0.6.0-0xenial-20180824-081412-0800_amd64.deb ...
Unpacking ros-kinetic-gencpp (0.6.0-0xenial-20180824-081412-0800) over (0.6.0-0xenial-20180809-135750-0800) ...
Preparing to unpack .../ros-kinetic-geneus_2.2.6-0xenial-20180824-081429-0800_amd64.deb ...
Unpacking ros-kinetic-geneus (2.2.6-0xenial-20180824-081429-0800) over (2.2.6-0xenial-20180809-135801-0800) ...
Preparing to unpack .../ros-kinetic-genlisp_0.4.16-0xenial-20180824-081624-0800_amd64.deb ...
Unpacking ros-kinetic-genlisp (0.4.16-0xenial-20180824-081624-0800) over (0.4.16-0xenial-20180809-135737-0800) ...
[... 422 lines cut ...]
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
[... cut ...]
Starting program: /opt/ros/kinetic/lib/joy/joy_node
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1856700 (LWP 1409)]
[New Thread 0x7ffff1055700 (LWP 1410)]
[New Thread 0x7ffff0854700 (LWP 1411)]
[New Thread 0x7fffebfff700 (LWP 1416)]
[ERROR] [1538315907.249739446]: Couldn't open joystick /dev/input/js0. Will retry every second.
```

Any chance to have some tests added that this is not occurring again?

stevensu1838, 2018-11-08:
Same error
ROS kinetic Ubuntu 16.04 Xbox360 wired controller
Followed tutorial: [http://wiki.ros.org/joy/Tutorials/ConfiguringALinuxJoystick]()

```
user@mech1349:~$ rosparam set joy_node/dev "/dev/input/js0"
user@mech1349:~$ rosrun joy joy_node
Segmentation fault (core dumped)
```
Can anyone help me out???

JWhitleyAStuff, 2018-11-19:
This was a known issue related to ros::Time in roscpp. It had to do with the order of operations on the ROS build farm. Please make sure your roscpp is updated to the latest version (`sudo apt update && sudo apt install ros-$ROS_DISTRO-roscpp`). If the issue is not resolved, please let me know ASAP and the versions of roscpp and joy that you are using. Thanks.

closed 2018-11-19 by JWhitleyAStuff

NateElm, 2018-11-20:
I am still seeing this with joy version 1.12.0 and roscpp version 1.12.14