ros-drivers/joystick_drivers #137: Before an axis event occurs on the joystick, the axis values published all default to zero

bucklandDD, 2018-11-18:
This results in potentially dangerous scenarios with certain controllers, which have axis values that go from -1 to 1. This means that 0 in that scale is 50%. 

Say an axis is mapped to throttle on a vehicle. Then if that "trigger" is not pressed the joystick driver will be publishing zero for that axis. Since the system maps zero to 50% on the axis scale, the vehicle will be throttling at 50% even before the pedal is pressed.

JWhitleyAStuff, 2018-11-19:
@bucklandDD - This was supposed to have been fixed by #91 which lets you set a param called `default_trig_val` to `true`. However, looking at the code, I don't think this is what this does at all. Can you verify and, if so, offer a PR with a solution?

JWhitleyAStuff, 2018-11-19:
May also be worth noting that, in applications where we have used `joy`, we have waited to receive a non-0 value before accepting input from the triggers.

closed 2018-11-30 by JWhitleyAStuff