ros-drivers/joystick_drivers #146: Couldn't open joystick force feedback!

mshong0320, 2019-06-27:
Trying to run the joy_node with logitech joystick controller on Kinetic version Ubuntu 16.04, but it's giving me an error [ERROR] [1561675023.851541705]: Couldn't open joystick force feedback!
Tried with multiple joy sticks and i can't figure out what's causing this and how to fix it.

JWhitleyAStuff, 2019-07-18:
This is just a warning. It is letting you know that force feedback is not available on the joystick that you are using. It does not affect normal use of the controller.

closed 2019-07-18 by JWhitleyAStuff

basvroeg, 2019-07-30:
if it's just warning, then also display it as a warning instead of an error please...

this tends to get confusing quickly!

nathangeorge1, 2019-11-11:
If possible, you can change ROS_ERROR to ROS_WARN if this is the case.

JWhitleyAStuff, 2019-11-11:
@nathangeorge1 / @basvroeg - PRs are welcome!

aaronaaronson98, 2020-11-26:
I just re-found this in my testing of the ros2 branch. I'm not sure how the PR process works on this repo (first timer!) but would this help?

https://github.com/ros-drivers/joystick_drivers/pull/189

PR #193 (2020-12-01): Change force feedback logger for dashing
Fix #146 for dashing by switching the ERROR to WARN.