ros-drivers/joystick_drivers #149: How to pair ps3 on TX2?

Tengyun-Mo, 2019-07-22:
I have a bluetooth adapters named CSR4.0 .The node didn't work when I paired PS3.

Tengyun-Mo, 2019-07-22:
My system is ubuntu18.04

Tengyun-Mo, 2019-07-22:
I can't found ps3 via device search.I don't where the problem is?Who can help me?I'm very appreciate it.

LazyEngineerToBe, 2019-07-26:
Can you see the device when you run `lsusb` ?

Tengyun-Mo, 2019-07-26:
Thanks for your response.Yes,I can see the device when I run lsusb.The bluetooth adapter can find other device like wireless keyboard,but it can not find ps3.

LazyEngineerToBe, 2019-07-26:
If you can't see the device listed in the terminal, then it's a hardware issue. Unrelated to ROS. Are you sure your controller is compatible with Bluetooth? It might be using a different RF protocol.

Tengyun-Mo, 2019-08-02:
Thanks,I changed another buletooth adapter,and it worked well.Thank you again

closed 2019-08-02 by JWhitleyAStuff