ros-drivers/joystick_drivers #153: Setting maximum speed when pressing a button

giuseppe-chiari, 2019-12-03:
Hi,
I have a robot that is controlled by a Logitech joystick through this package. I was wondering if it was possible to set the buttons 1,2,3 of the joystick in order to set a maximum speed so that when one of those buttons is pressed the robot will have a top speed and won't go any faster.
For example:
-button 1: when pressed the maximum speed is 1 mph/h
-button 2: when pressed the maximum speed is 2 mph/h
-button 3: when pressed the maximum speed is 3 mph/h

JWhitleyAStuff, 2019-12-03:
This is something that would be handled at a higher level than the `joy` node. The `joy` node is only intended to handle the raw input from the hardware and present a `sensor_msgs/Joy` message reporting it. You would need a node that receives that message and decides what to do based on the button pressed.

closed 2019-12-03 by JWhitleyAStuff

giuseppe-chiari, 2019-12-04:
@JWhitleyAStuff  thank you for the explanation! I think [this](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Joy.html) is the documentation of `sensor_msgs/Joy`. I see that the message is made by two main components: `axes` and `buttons`. And since I'm interested in which button has what value I need to look into the array called `buttons`. Where can I get a more specific documentation about how this array is filled with data? What I mean is: when a button is pressed an int32 will be inserted into the array, but which int32 specifically? Or is it just a true/false situation where true is 1 and false is 0?

JWhitleyAStuff, 2019-12-04:
@giuseppe-chiari - The implementation depends on how your operating system interprets the inputs from that specific controller. In general, if it is just a momentary switch (like most X/Y/A/B or triangle/square/circle/X buttons), the values in the buttons array will only be 0 or 1. However, the order of the buttons array is controller-dependent.