ros-drivers/joystick_drivers #163: No frame_id in the header of the joystick msg (joy_node)

MamounG, 2020-04-07:
If multiple joysticks are used, they all publish in the same topic, with no frame_id, which might introduce errors.

simple fix: adding the joy_dev_ as frame_id:
in line 518 of joystick_drivers/joy/src/joy_node.cpp add:
`joy_msg.header.frame_id = joy_dev_.c_str();`

JWhitleyWork, 2020-04-07:
@MamounG Would you mind creating a merge request to resolve this isssue?

MamounG, 2020-04-08:
I can do that but this is going to be my first contribution in github, so I am not sure of how to do:
I cannot manage to push a new branch, is it a permission problem? or is it me doing something wrong?
cheers!

JWhitleyWork, 2020-04-09:
@MamounG Not a problem! So, Github uses what's called a "fork-and-pull" model. You create a fork of the project, then create a "Pull Request" to include your changes to the original. Here's a good overview: https://reflectoring.io/github-fork-and-pull/ Let me know if you have any questions.

PR #166 (2020-04-15): frame_id in the header of the joystick msg
this PR closes #163
closed 2020-04-15 by JWhitleyWork