ros-drivers/joystick_drivers #165: ROS 2: Create Chocolatey Package for SDL2 to Replace sdl2_vendor

JWhitleyWork, 2020-04-10:
Currently no package exists for Windows for SDL2 which can be resolved by `rosdep`. Pretty sure Chocolatey is the dependency resolution framework for ROS 2 on Windows (@clalancette can you confirm?) so someone should produce an SDL2 package for it so we don't have to keep the wrapper package around for just one platform.

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): The ROS2 `ros2` branch still declares `sdl2_vendor` from `joy/package.xml` and CMake. Merged PR [#157](https://github.com/ros-drivers/joystick_drivers/pull/157) added the vendor package as part of cross-platform SDL support; its discussion records the attempt to use the system `sdl2` rosdep key and why the wrapper was retained. PR [#270](https://github.com/ros-drivers/joystick_drivers/pull/270) only upgraded the SDL source version to 2.0.20.

I found no merged Chocolatey SDL2 package or replacement of `sdl2_vendor`; the requested Windows dependency packaging remains outstanding.

Source: [audited code/configuration](https://github.com/ros-drivers/joystick_drivers/blob/50bcf0f44d40368201aec8f451227af384429c50/joy/package.xml#L23-L28).

clalancette, 2026-11-16:
We've now switched away from choclatey to pixi/conda, which may have the dependency available.