ros-drivers/joystick_drivers #170: Release joystick_drivers into ROS Noetic?

sloretz, 2020-04-23:
Would you mind releasing `joystick_drivers` to ROS Noetic? It looks like all of its dependencies have been released, and recursively 33 repos need it.

I'm not sure how much work it will need. The [Noetic Migration Guide](http://wiki.ros.org/noetic/Migration) or the [guide about transitioning ROS packages to Python 3](http://wiki.ros.org/UsingPython3/) may be useful.

doisyg, 2020-05-26:
It compiles fine as long as the following debs are installed (don't know if rosdep resolves them): 
libusb-dev libx11-dev libspnav-dev libbluetooth-dev libcwiid-dev
Would love to see it released too

JWhitleyWork, 2020-06-02:
@sloretz or @doisyg - I have updated `master` to target Noetic. Could either or both of you provide some testing prior to me creating a release?

BPetitSurble, 2020-06-02:
Hello @JWhitleyWork, thank you for the port!

I tried to test by building in the docker image you use in your ci and running `roslaunch joy ps4joy.launch` I got the error:

```
process[joy_remap-3]: started with pid [1774]
/usr/bin/env: ‘python’: No such file or directory
[joy_remap-3] process has died [pid 1774, exit code 127, cmd /opt/workspace/src/joystick_drivers/joy/scripts/joy_remap.py joy_in:=joy_orig joy_out:=joy __name:=joy_remap __log:=/root/.ros/log/dfcd11a0-a4b0-11ea-b82c-0242ac110002/joy_remap-3.log].
log file: /root/.ros/log/dfcd11a0-a4b0-11ea-b82c-0242ac110002/joy_remap-3*.log
```
May it be due to a migration to python3 issue?

By the way it looks like it is possible to keep using catkin_tools like in kinetic-devel and melodic-devel using the workaround mentioned in [this issue](https://github.com/catkin/catkin_tools/issues/594#issuecomment-635621597).

I could do it by replacing `apt-get update -qq && apt-get install -y python-catkin-tools` with:
```
apt-get update -qq && apt-get install -y curl git
curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
python3  get-pip.py && rm get-pip.py
pip install -U "git+https://github.com/catkin/catkin_tools.git#egg=catkin_tools"
```
Of course it may not be the best solution as it does not rely on a released version of their software, just thought it was worth mentioning.

doisyg, 2020-06-02:
`joy_node` working fine on my side

NathanDodson, 2020-06-23:
Any update on the Noetic release of this package?

rhaschke, 2020-07-01:
Ping. The missing Noetic release is blocking the MoveIt release.

JWhitleyWork, 2020-07-07:
@rhaschke @sloretz Sorry for the long delay. PR is in. https://github.com/ros/rosdistro/pull/25680

closed 2020-07-07 by JWhitleyWork