ros-drivers/joystick_drivers #184: How to utilize FF_CONSTANT option for force feedback.

JeyP4, 2020-08-31:
Hello

I am not able to utilize FF_CONSTANT force effect. My try code is:
```
if (iwantconstantforce)
        {
          joy_effect_.id = -1;
          joy_effect_.type = FF_CONSTANT;
          joy_effect_.direction = 0xC000;  // down
          joy_effect_.replay.length = 100;
          joy_effect_.replay.delay = 0;

          joy_effect_.trigger.button = 0;
          joy_effect_.trigger.interval = 100;

          joy_effect_.u.constant.level = 65535;
          joy_effect_.u.constant.envelope.attack_length = joy_effect_.replay.length / 10;
          joy_effect_.u.constant.envelope.fade_length = joy_effect_.replay.length / 10;

          joy_effect_.u.constant.envelope.attack_level = joy_effect_.u.constant.level / 10;
          joy_effect_.u.constant.envelope.fade_level = joy_effect_.u.constant.level / 10;
        }

```
I am able to produce FF_SPRING and FF_DAMPER effects with following codes.
```
        if (youwantdampereffect)
        {
          joy_effect_.id = -1;
          joy_effect_.direction = 0;  // down
          joy_effect_.type = FF_DAMPER;
          joy_effect_.replay.length = 20;
          joy_effect_.replay.delay = 0;
          
          joy_effect_.u.condition[0].right_saturation = 65535;
          joy_effect_.u.condition[0].left_saturation = 65535;
          joy_effect_.u.condition[0].right_coeff = 65535 / 2;
          joy_effect_.u.condition[0].left_coeff = 65535 / 2;
          joy_effect_.u.condition[0].deadband = 0;
          joy_effect_.u.condition[0].center = 0;

          int ret = ioctl(ff_fd_, EVIOCSFF, &joy_effect_);        // upload the effect
        }

        if (youwantspringeffect)
        {
          joy_effect_2.id = -1;
          joy_effect_2.direction = 0;  // down
          joy_effect_2.type = FF_SPRING;
          joy_effect_2.replay.length = 20;
          joy_effect_2.replay.delay = 0;

          joy_effect_2.u.condition[0].right_saturation = 65535 / 2;
          joy_effect_2.u.condition[0].left_saturation = 65535 / 2;
          joy_effect_2.u.condition[0].right_coeff = 32767;
          joy_effect_2.u.condition[0].left_coeff = 32767;
          joy_effect_2.u.condition[0].deadband = 0;
          joy_effect_2.u.condition[0].center = 0;

          int ret = ioctl(ff_fd_, EVIOCSFF, &joy_effect_2);        // upload the effect
        }
```

I do not find any info about what is constant force effect feels like or when it makes sense to use it.
Can somebody brief its importance and usage?

Thanks :)

JWhitleyWork, 2020-09-01:
@JeyP4 I'm not sure where you are getting `FF_CONSTANT`. This is not part of `joystick_drivers` nor part of the [JoyFeedback](http://docs.ros.org/noetic/api/sensor_msgs/html/msg/JoyFeedback.html) message.

JeyP4, 2020-09-01:
You are right.
It is a one of the feature available in [Force feedback for Linux](https://www.kernel.org/doc/html/v4.15/input/ff.html#instructions-to-the-developer) like FF_RUMBLE mentioned in [joy/src/joy_node.cpp](https://github.com/ros-drivers/joystick_drivers/blob/main/joy/src/joy_node.cpp#L281).

So my concern is, if someone knows how to use FF_CONSTANT instead of FF_RUMBLE !!

JWhitleyWork, 2020-09-02:
Unfortunately, this isn't directly related to `joystick_drivers` so I'm gonna close it. I recommend trying Stack Overflow unless you think it is more ROS-related, in which case I'd say ROS Answers.

closed 2020-09-02 by JWhitleyWork