ros-drivers/joystick_drivers #200: Problem starting ps3joy.py node on Ubuntu 20.04/Noetic

lubaroli, 2021-02-16:
Hello, 

I'm trying to run ps3joy on Ubuntu 20.04 and after successfully installing ros-noetic-joy, ros-noetic-joystick-drivers, and pairing the remote via USB I get the following error when starting the node:
```
user@localhost: sudo bash
root@localhost: source /opt/ros/noetic/setup.bash
root@localhost: rosrun ps3joy ps3joy.py

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/ps3joy/ps3joy.py", line 436, in <module>
    cm = connection_manager(decoder(inactivity_timeout=inactivity_timeout,
  File "/opt/ros/noetic/lib/ps3joy/ps3joy.py", line 169, in __init__
    self.joy = uinputjoy(buttons, axes, axmin, axmax, axfuzz, axflat)
  File "/opt/ros/noetic/lib/ps3joy/ps3joy.py", line 104, in __init__
    struct.pack(uinput_user_dev, "Sony Playstation SixAxis/DS3",
struct.error: argument for 's' must be a bytes object
```

Is this package compatible with python 3.8?

mikaelarguedas, 2021-03-06:
+1 there seem to be multiple issues with this package's in Noetic.
Based on commit history: @JWhitleyWork @clalancette do you know if this package has been ported to Python3 prior to being released in Noetic ?
Is there ongoing effort to make it Python3 compatible ?

/cc @sloretz

clalancette, 2021-03-06:
> Is there ongoing effort to make it Python3 compatible ?

Not as far as I know.  I'm happy to review PRs for it.

mikaelarguedas, 2021-03-08:
> Not as far as I know. I'm happy to review PRs for it.

Sounds good, in the meantime is it possible to remove the deb from the repos ?
It is pretty misleading for people doing due diligence of packages available to see a deb available but then find out the packages are installable but not usable.

clalancette, 2021-03-08:
> Sounds good, in the meantime is it possible to remove the deb from the repos ?

I'll defer judgement of that to @JWhitleyWork who does most of the maintenance here.  @sloretz , opinion here is also welcome.

JWhitleyWork, 2021-03-08:
@clalancette @mikaelarguedas Thanks for bringing this up. I've been pretty neglectful of this repo for a while and it has obviously caused some problems. However, I hate to pull the packages from the build farm in this staack that have either little or no Python dependency (like `joy`) that people rely on heavily. To that end:

- `joy`: Only two Python scripts:
  - `test/test_joy_msg_migration.py` - This is obsolete because the `sensor_msgs::Joy` message hasn't changed since before Indigo
  - `scripts/joy_remap.py` - Seems to be fine in Python3.8
- `joystick_drivers` - meta-package - no changes required
- `ps3joy` - Obviously has problems, as pointed out in this issue
- `spacenav_node` - pure C++ - no issues
- `wiimote` - Pure Python package - needs port

Based on the above, I think we should temporarily remove `ps3joy` and `wiimote` from the build farm until they have been ported and verified. Additionally, it looks like there may be problems with `ps3joy` on the ROS2 branches as well since the code looks basically identical to the Noetic version other than ros2-specific changes. I've opened #204 to track the porting effort. Thoughts? Feedback?

clalancette, 2021-03-08:
@JWhitleyWork Actually, the situation on ROS 2 is simple.  `ps3joy` is explicitly disabled via an `AMENT_IGNORE` file there, so no work needs to be done.

On Noetic, the situation is actually a bit more complex.  `ps3joy` clearly can't work in a ROS environment without a port to Python3.  However, as you correctly point out, `wiimote` actually has both a Python3 and a C++ version.  In theory the C++ version works in Noetic, though I haven't tested it one way or the other.  I think we should just leave `wiimote` around in the chance that someone is already using the C++ version.

Thus, I think the only thing that we should do here is to disable `ps3joy` on Noetic.

mikaelarguedas, 2021-03-10:
Thanks for the thorough answer. Removing the ps3joy deb from the repo until the package is ported to Python3 sounds good to me

mikaelarguedas, 2021-06-07:
Any update on this issue ? looks like the ps3joy Noetic deb is still up, pulling python2 dependencies and not running with Python3

JWhitleyWork, 2021-06-07:
I'm not sure how I was able to release `wiimote` into `noetic` before but it appears that it is now missing a required package (`cwiid-dev`) in Fedora so I had to add a `CATKIN_IGNORE` to that package as well. `joy`, `joystick_drivers`, and `spacenav_node` are now being released as 1.15.1. Here's the PR: https://github.com/ros/rosdistro/pull/29793