ros-drivers/joystick_drivers #208: ROS2 install not working on RPI Ubuntu 20.04

somervda, 2021-04-02:
I have done two installs , one on a Raspberry PI 4 running Ubuntu 20.04 and an identical ROS2 install on a i386 machine. The joy_node is working on the i386 ROS2 but does not seem to connect to the joystick on the RPI . I am using the same joystick controller.

I went though the process of installing the joy package

>  sudo apt install ros-foxy-joy

And could see my joystick active, on both installs, on /dev/input/js0 using a 

>   jstest /dev/input/js0 

and run the joy_node

> ros2 run joy joy_node --ros-args -p device_name:=/dev/input/js0

On the i386 install the joy_node reports see in the joystick and reports as well as publishing /joy messages

> [INFO] [1617369568.897123043] [joy_node]: Opened joystick: Microsoft X-Box 360 pad.  deadzone: 0.050000

On the RPI I dont see any [INFO] message from the joy_node and no /joy messages are produced.

BTW: Im guessing the problem may be at the os level on the i386 the lsusb looks like this

> Bus 002 Device 003: ID 0bda:0138 Realtek Semiconductor Corp. RTS5138 Card Reader Controller
Bus 002 Device 002: ID 0402:9665 ALi Corp. Gateway Webcam
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 003 Device 004: ID 045e:028e Microsoft Corp. Xbox360 Controller
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

on the RPI it looks like this (joystick details are not being identified/displayed for device #8)

> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 2563:0575
Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

ShotaAk, 2021-04-27:
I've encountered the same problem.

How about the `joy_linux_node`? It worked in my environment.

https://github.com/ros-drivers/joystick_drivers/tree/ros2/joy_linux

somervda, 2021-04-27:
Thanks - that worked

closed 2021-04-27 by somervda