ros-drivers/joystick_drivers #221: spacenav_node not working

BhanuKiranChaluvadi, 2021-09-28:
Hi, 

Using ubuntu 20.04, ros-foxy, I Directly Installed space-nav `sudo apt install ros-foxy-spacenav`
While running the spacenav_node `$ ros2 run spacenav spacenav_node`
```
terminate called after throwing an instance of 'class_loader::LibraryLoadException'
  what():  Could not load library LoadLibrary error: libspacenav.so: cannot open shared object file: No such file or directory, at /tmp/binarydeb/ros-foxy-rcutils-1.1.3/src/shared_library.c:84
```

`ros-foxy-rcutils` is already installed on my system

```
sudo apt install ros-foxy-rcutils
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-foxy-rcutils is already the newest version (1.1.3-1focal.20210901.035300).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

I also tried tried compiling directly from source, the colcon build is succeed but the when i run the node, it has same error as   before

@jbohren

MeticulousFishFrier, 2021-11-04:
@adrianramospeon's suggestion works. Perhaps you should submit a pull request for this fix.

alfaDoG1993, 2022-02-11:
This trick tested and works on ros2 galactic too

JWhitleyWork, 2022-02-14:
Closed by #229.

closed 2022-02-14 by JWhitleyWork

gedaliahknizhnik, 2023-03-09:
I installed spacenav for ROS2 Foxy via APT and still had to do the file copy in order to get the node to run. So it seems the fix didn't take

commit 85c6b56a (2022-02-14): Installing libspacenav.so to lib/ for spacenav_node execution via ros2 run (#229)

Fixed #221
commit 8f27d12f (2022-02-14): Installing libspacenav.so to lib/ for spacenav_node execution via ros2 run (#229)

Fixed #221