ros-drivers/joystick_drivers #223: ROS2 spacenav_node doesn't publish button values on to /spacenav/joy

MeticulousFishFrier, 2021-11-04:
After fixing my ROS2 spacenav_node install with [the fix in issue #221](https://github.com/ros-drivers/joystick_drivers/issues/221#issuecomment-942475652), I noticed that the button values of the space mouse are not published on to the /spacenav/joy topic. Attached below is an example echo of the topic. All the axes values are correct, but no button values are shown. 

[image]

stefanscherzinger, 2022-11-15:
I ran into this a few days ago. The buttons are actually published, but the default frequency of `1Khz` floods them in the terminal. [Here's a fix](https://github.com/stefanscherzinger/joystick_drivers/tree/fix-buttons) that restores the previous ROS1 behavior.

JWhitleyWork, 2022-12-04:
@MeticulousFishFrier Please test out the new behavior and re-open this issue if it is not resolved.

PR #243 (2023-01-07): Fix publishing of spacenav button values
## Problem
The spacenav buttons are currently only published once upon being triggered. They are thus flooded by the other, continuously published motion events.

## Approach

We now always publish the two default button states, in which
0 = not pressed
1 = pressed.

That's the previously known ROS1 behavior.

Additional buttons emerge and stay in that list upon being pressed for the first time.  This supports more advanced spacenav devices with a yet unknown number of buttons.

Fixes #223

---

## What the fix looks like
`ros2 topic echo /spacenav/joy` [... cut]
closed 2022-12-04 by JWhitleyWork