ros-drivers/joystick_drivers #226: spacenav build failure in ROS 2 Rolling

nuclearsandwich, 2022-01-17:
* Package: `spacenav`
* Version: 3.0.0-4
* ROS distro: `rolling`
* Platforms
  * [Ubuntu AMD64](https://build.ros2.org/job/Rbin_uF64__spacenav__ubuntu_focal_amd64__binary)
  * [Ubuntu ARM64](https://build.ros2.org/job/Rbin_ufv8_uFv8__spacenav__ubuntu_focal_arm64__binary)

Since previous builds of this package succeeded it is probable that this build error was introduced by a change in a dependency.

Here's an excerpt from the most recent failure on AMD64](https://build.ros2.org/job/Rbin_uF64__spacenav__ubuntu_focal_amd64__binary/86/):

```
[ 75%] Building CXX object CMakeFiles/spacenav.dir/src/spacenav.cpp.o
/usr/lib/ccache/c++  -DDEFAULT_RMW_IMPLEMENTATION=rmw_cyclonedds_cpp -DRCUTILS_ENABLE_FAULT_INJECTION -Dspacenav_EXPORTS -I/tmp/binarydeb/ros-rolling-spacenav-3.0.0/include -isystem /opt/ros/rolling/include  -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-rolling-spacenav-3.0.0=. -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC   -Wall -Wextra -Wpedantic -std=gnu++17 -o CMakeFiles/spacenav.dir/src/spacenav.cpp.o -c /tmp/binarydeb/ros-rolling-spacenav-3.0.0/src/spacenav.cpp
In file included from /tmp/binarydeb/ros-rolling-spacenav-3.0.0/src/spacenav.cpp:37:
/opt/ros/rolling/include/rclcpp_components/register_node_macro.hpp:18:10: fatal error: class_loader/class_loader.hpp: No such file or directory
   18 | #include "class_loader/class_loader.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [CMakeFiles/spacenav.dir/build.make:66: CMakeFiles/spacenav.dir/src/spacenav.cpp.o] Error 1
make[4]: Leaving directory '/tmp/binarydeb/ros-rolling-spacenav-3.0.0/obj-x86_64-linux-gnu'
make[3]: *** [CMakeFiles/Makefile2:110: CMakeFiles/spacenav.dir/all] Error 2
make[3]: Leaving directory '/tmp/binarydeb/ros-rolling-spacenav-3.0.0/obj-x86_64-linux-gnu'
make[2]: *** [Makefile:144: all] Error 2
make[2]: Leaving directory '/tmp/binarydeb/ros-rolling-spacenav-3.0.0/obj-x86_64-linux-gnu'
```

JWhitleyWork, 2022-01-31:
@nuclearsandwich This appears to be fixed. Please re-open if it is still an issue.

closed 2022-01-31 by JWhitleyWork

nuclearsandwich, 2022-01-31:
Indeed it was fixed with https://github.com/ros-drivers/joystick_drivers/pull/228