ros-drivers/joystick_drivers #231: ROS2: Multiple Joysticks are not working

whoenig, 2022-02-15:
When using the `foxy_devel` (or `ros2`) branch, multiple joysticks are not working. In particular

```
ros2 run joy joy_node --ros-args -p device_id:=0
```
works as expected, but

```
ros2 run joy joy_node --ros-args -p device_id:=1
```
never opens the appropriate second joystick.

The root cause is that SDL is re-initialized in a loop, just after checking for a single event. Multiple joysticks will create multiple events (one for each joystick), however. A fix is to initialize/cleanup SDL simply once, see https://github.com/IMRCLab/joystick_drivers/tree/bugfix_multiple_joysticks.

Noel215, 2023-05-30:
Hi, this issue is also happening to me.
Is there any intention to fix it?

clalancette, 2023-09-14:
I believe this should be fixed now (by #247), so I'll close this out.

If you are still having problems, please feel free to reopen or open a new bug.

closed 2023-09-14 by clalancette