ros-drivers/joystick_drivers #241: Humble deb package attempts to uninstall other important packages.

roni-kreinin, 2022-11-08:
When I attempt to install `ros-humble-joy` on a RPi4 running Ubuntu 22.04 server, apt attempts to uninstall a bunch of important packages:

```
$ sudo apt install ros-humble-joy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  adwaita-icon-theme bolt busybox-initramfs cryptsetup cryptsetup-initramfs devio dns-root-data dnsmasq-base finalrd fwupd-signed gdal-data gir1.2-packagekitglib-1.0 gstreamer1.0-plugins-base
  gtk-update-icon-cache hicolor-icon-theme humanity-icon-theme ibverbs-providers initramfs-tools-bin klibc-utils libaacs0 libaec0 libaom3 libappstream4 libarmadillo10 libarpack2 libatasmart4
  libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3 libavahi-common-data libavahi-common3 libavcodec-dev libavcodec58 libavformat-dev libavformat58 libavutil-dev libavutil56
  libbdplus0 libblockdev-crypto2 libblockdev-fs2 libblockdev-loop2 libblockdev-part-err2 libblockdev-part2 libblockdev-swap2 libblockdev-utils2 libblockdev2 libblosc1 libbluetooth3 libbluray2
  libboost-dev libboost-python1.74.0 libboost1.74-dev libcairo-gobject2 libcdparanoia0 libcfitsio9 libcharls2 libchromaprint1 libcodec2-1.0 libcolord2 libcups2 libdav1d5 libdc1394-25 libdc1394-dev
  libdconf1 libde265-0 libdeflate-dev libdouble-conversion3 libdw1 libefiboot1 libefivar1 libepoxy0 libevent-pthreads-2.1-7 libexif-dev libexif-doc libexif12 libfabric1 libflashrom1 libfreexl1
  libftdi1-2 libfwupd2 libfwupdplugin5 libfyba0 libgcab-1.0-0 libgdal30 libgdcm-dev libgdcm3.0 libgeos-c1v5 libgeos3.10.2 libgeotiff5 libgif7 libgl2ps1.4 libglew2.2 libgme0 libgpgme11 libgphoto2-6
  libgphoto2-dev libgphoto2-l10n libgphoto2-port12 libgsm1 libgstreamer-plugins-base1.0-0 libgstreamer1.0-0 libgudev-1.0-0 libgusb2 libhdf4-0-alt libhdf5-103-1 libhdf5-hl-100 libheif1 libhwloc-plugins
  libhwloc15 libibverbs1 libilmbase-dev libilmbase25 libiniparser1 libisns0 libjbig-dev libjcat1 libjpeg-dev libjpeg-turbo8-dev libjpeg8-dev libjson-glib-1.0-0 libjson-glib-1.0-common libklibc
  libkmlbase1 libkmldom1 libkmlengine1 liblept5 liblzma-dev libmbim-glib4 libmbim-proxy libminizip1 libmm-glib0 libmp3lame0 libmpg123-0 libmysqlclient21 libndp0 libnetcdf19 libnetplan0 libnm0 libnorm1
  libnspr4 libnss3 libodbc2 libodbcinst2 libogdi4.1 libopencv-calib3d4.5d libopencv-core-dev libopencv-core4.5d libopencv-dnn-dev libopencv-dnn4.5d libopencv-features2d4.5d libopencv-flann-dev
  libopencv-flann4.5d libopencv-imgcodecs-dev libopencv-imgcodecs4.5d libopencv-imgproc-dev libopencv-imgproc4.5d libopencv-ml-dev libopencv-ml4.5d libopencv-objdetect4.5d libopencv-photo-dev
  libopencv-photo4.5d libopencv-shape-dev libopencv-shape4.5d libopencv-stitching4.5d libopencv-video-dev libopencv-video4.5d libopencv-videoio-dev libopencv-videoio4.5d libopencv-videostab4.5d
  libopencv-viz-dev libopencv-viz4.5d libopenexr-dev libopenexr25 libopeniscsiusr libopenmpi3 libopenmpt0 liborc-0.4-0 libpackagekit-glib2-18 libparted-fs-resize0 libpgm-5.3-0 libplymouth5 libpmix2
  libpng-dev libpng-tools libpolkit-agent-1-0 libpolkit-gobject-1-0 libpoppler118 libpq5 libproj22 libprotobuf23 libqhull-r8.0 libqmi-glib5 libqmi-proxy librabbitmq4 libraw1394-11 libraw1394-dev
  libraw1394-tools librdmacm1 librsvg2-2 librsvg2-common librttopo1 libsgutils2-2 libshine3 libsnappy1v5 libsocket++1 libsoxr0 libspatialite7 libspeex1 libsrt1.4-gnutls libssh-gcrypt-4 libstemmer0d
  libsuperlu5 libswresample-dev libswresample3 libswscale-dev libswscale5 libsz2 libtbb-dev libtbb12 libtbb2 libtbbmalloc2 libtcl8.6 libteamdctl0 libtesseract4 libtheora0 libtiff-dev libtiffxx5
  libtk8.6 libtwolame0 libubootenv-tool libubootenv0.1 libucx0 libudfread0 libudisks2-0 libunwind8 liburcu8 liburiparser1 libusb-1.0-0-dev libusb-1.0-doc libva-drm2 libva-x11-2 libva2 libvdpau1
  libvisual-0.4-0 libvolume-key1 libvorbisfile3 libvpx7 libvtk9.1 libx264-163 libx265-199 libxcomposite1 libxdamage1 libxerces-c3.2 libxft2 libxmlb2 libxnvctrl0 libxtst6 libxvidcore4 libzmq5
  libzvbi-common libzvbi0 lz4 mesa-va-drivers mesa-vdpau-drivers mtd-utils mysql-common nlohmann-json3-dev ocl-icd-libopencl1 opencv-data poppler-data ppp pptp-linux proj-bin proj-data
  python3-software-properties ros-humble-camera-calibration-parsers ros-humble-camera-info-manager ros-humble-depthai-ros-msgs ros-humble-foxglove-msgs ros-humble-image-transport ros-humble-vision-msgs
  ros-humble-xacro sbsigntool secureboot-db session-migration sg3-utils squashfs-tools tcl tcl8.6 u-boot-tools ubuntu-mono unixodbc-common usb-modeswitch usb-modeswitch-data va-driver-all
  vdpau-driver-all
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  alsa-topology-conf alsa-ucm-conf cryptsetup cryptsetup-initramfs dbus gir1.2-ibus-1.0 libasound2 libasound2-data libasound2-dev libasyncns0 libblkid-dev libdbus-1-3 libdbus-1-dev libdecor-0-0
  libdecor-0-dev libdecor-0-plugin-1-cairo libdrm-dev libdrm-etnaviv1 libdrm-freedreno1 libdrm-tegra0 libegl-dev libegl-mesa0 libegl1 libegl1-mesa-dev libffi-dev libflac8 libfwupd2 libfwupdplugin5
  libgbm-dev libgbm1 libgl-dev libgles-dev libgles1 libgles2 libglib2.0-dev libglib2.0-dev-bin libglu1-mesa libglu1-mesa-dev libglvnd-core-dev libglvnd-dev libglx-dev libibus-1.0-5 libibus-1.0-dev
  libice-dev libmount-dev libopengl-dev libpciaccess-dev libpcre16-3 libpcre2-16-0 libpcre2-32-0 libpcre2-8-0 libpcre2-dev libpcre2-posix3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpthread-stubs0-dev
  libpulse-dev libpulse-mainloop-glib0 libpulse0 libsdl2-2.0-0 libsdl2-dev libselinux1-dev libsepol-dev libsm-dev libsndfile1 libsndio-dev libsndio7.0 libudev-dev libudev1 libwayland-bin libwayland-dev
  libwayland-server0 libx11-dev libxau-dev libxcb1-dev libxcursor-dev libxdmcp-dev libxext-dev libxfixes-dev libxi-dev libxinerama-dev libxkbcommon-dev libxrandr-dev libxrender-dev libxss-dev libxt-dev
  libxv-dev libxv1 libxxf86vm-dev ros-humble-sdl2-vendor tiny-initramfs tiny-initramfs-core uuid-dev x11proto-dev xorg-sgml-doctools xtrans-dev
Suggested packages:
  keyutils default-dbus-session-bus | dbus-session-bus libasound2-plugins alsa-utils libasound2-doc libgirepository1.0-dev libglib2.0-doc libice-doc pulseaudio xdg-utils libsm-doc sndiod libwayland-doc
  libx11-doc libxcb-doc libxext-doc libxt-doc
Recommended packages:
  plymouth
The following packages will be REMOVED:
  at-spi2-core bluez cloud-init cloud-initramfs-copymods cloud-initramfs-dyn-netconf dbus-user-session dconf-gsettings-backend dconf-service flash-kernel friendly-recovery fwupd
  gsettings-desktop-schemas initramfs-tools initramfs-tools-core kpartx libgtk-3-0 libgtk-3-bin libgtk-3-common libnss-systemd libopencv-calib3d-dev libopencv-contrib-dev libopencv-contrib4.5d
  libopencv-dev libopencv-features2d-dev libopencv-highgui-dev libopencv-highgui4.5d libopencv-objdetect-dev libopencv-stitching-dev libopencv-superres-dev libopencv-superres4.5d
  libopencv-videostab-dev libopencv4.5-java libopencv4.5d-jni libpam-systemd libtss2-esys-3.0.2-0 libtss2-mu0 libtss2-sys1 libtss2-tcti-cmd0 libtss2-tcti-device0 libtss2-tcti-mssim0 libtss2-tcti-swtpm0
  mdadm modemmanager multipath-tools netplan.io network-manager network-manager-pptp notification-daemon open-iscsi overlayroot packagekit packagekit-tools pi-bluetooth pkexec plymouth
  plymouth-theme-ubuntu-text policykit-1 polkitd python3-opencv ros-humble-cv-bridge ros-humble-depth-image-proc ros-humble-depthai ros-humble-depthai-bridge ros-humble-depthai-examples
  ros-humble-depthai-ros ros-humble-image-geometry sg3-utils-udev snapd software-properties-common tpm-udev ubuntu-minimal ubuntu-raspi-settings ubuntu-server ubuntu-server-raspi ubuntu-standard udev
  udisks2
The following NEW packages will be installed:
  alsa-topology-conf alsa-ucm-conf gir1.2-ibus-1.0 libasound2 libasound2-data libasound2-dev libasyncns0 libblkid-dev libdbus-1-dev libdecor-0-0 libdecor-0-dev libdecor-0-plugin-1-cairo libdrm-dev
  libdrm-etnaviv1 libdrm-freedreno1 libdrm-tegra0 libegl-dev libegl-mesa0 libegl1 libegl1-mesa-dev libffi-dev libflac8 libgbm-dev libgbm1 libgl-dev libgles-dev libgles1 libgles2 libglib2.0-dev
  libglib2.0-dev-bin libglu1-mesa libglu1-mesa-dev libglvnd-core-dev libglvnd-dev libglx-dev libibus-1.0-5 libibus-1.0-dev libice-dev libmount-dev libopengl-dev libpciaccess-dev libpcre16-3
[... cut ...]
  tiny-initramfs tiny-initramfs-core uuid-dev x11proto-dev xorg-sgml-doctools xtrans-dev
The following packages will be upgraded:
  cryptsetup cryptsetup-initramfs dbus libdbus-1-3 libfwupd2 libfwupdplugin5 libpcre2-8-0 libudev1
8 upgraded, 90 newly installed, 77 to remove and 66 not upgraded.
Need to get 13.9 MB of archives.
After this operation, 263 MB disk space will be freed.
Do you want to continue? [Y/n] 
```

Tested on an amd64 VM and had no issues, so it might be a problem with the arm64 package.

tfoote, 2022-11-11:
This looks like you might be out of date an hitting the bug in apt we traced down here: https://github.com/ros2/ros2/issues/1272

You should make sure you have updated apt before trying to do the installation.

roni-kreinin, 2022-11-14:
Running `sudo apt update && sudo apt upgrade` before installing Humble fixed the issue.