ros-drivers/joystick_drivers #245: How to Install spacenav

Akumar201, 2022-11-28:
Hello, I am trying to install spacenav_node and for that I have to install [spacenav](http://wiki.ros.org/spacenav#Building_SpaceNav), I would like to know how can I install this? I tried to follow the instruction on the ros webpage but it is not working for me. 

I am cloning the repository from github and inside it running the `rosmake spacenav `command. I am new to ROS , any help would be much appreciated.

JWhitleyWork, 2022-12-04:
Please provide the following:

- Linux distro and version
- ROS version
- Version of `joystick_drivers` you are trying to install
- Version of `spacenav` you are trying to install

Akumar201, 2022-12-05:
I am using Ubuntu 20.04
ROS Noetic
I am trying to install spacenav from here http://wiki.ros.org/spacenav_node

JWhitleyWork, 2024-02-23:
On Ubuntu, you can just `sudo apt install libspnav-dev`. However, this will be done automatically using `rosdep`. From the base of your workspace, run `sudo apt update && rosdep install --from-paths src -i -y` to get dependencies automatically.

closed 2024-02-23 by JWhitleyWork