ros-drivers/joystick_drivers #246: joy_node uses a lot of CPU when no joystick is connected

tonynajjar, 2022-12-06:
We have joy_node always running but the joystick is only connected in need to teleop the robot. I noticed that when it is not connected and the node is launched, it uses a lot of CPU. I tracked down the high CPU consumption to this [line](https://github.com/ros-drivers/joystick_drivers/blob/ros2/joy/src/joy.cpp#L479).

Why do we we need to call SDL_Quit in this loop? Is there a better way to avoid the high CPU consumption?

tonynajjar, 2022-12-06:
Also, we noticed that this is not a problem on galactic and given that the joy_node barely changed between galactic in Humble, maybe SDL changed

doisyg, 2023-02-23:
Noticing the same on  `Humble`

tonynajjar, 2023-02-23:
Can you check if the linked PR fixes it for you @doisyg

doisyg, 2023-06-12:
Sorry for the late answer @tonynajjar, confirming your PR fixes the issue for me

JWhitleyWork, 2023-06-16:
Fixed by #247 .

closed 2023-06-16 by JWhitleyWork

doisyg, 2023-06-21:
Can we release it also ? 🙏

JWhitleyWork, 2023-06-22:
> Can we release it also ? pray

Working on it. I'm currently missing some required access. See https://github.com/ros2-gbp/joystick_drivers-release/issues/9 for details.