ros-drivers/joystick_drivers #267: SDL_GameControllerHasRumble is undefine

tienake, 2023-09-26:
In src/game_controller.cpp line 324
SDL_GameControllerHasRumble is undefine
Fail to build joy

mor2code, 2023-10-02:
focal 20.04  SDL version is 20.10, it does not have SDL_GameControllerHasRumble function was implemented at SDL 2.18.

I have replaced the call with SDL_GameControllerRumble sending zeros in all 3 parameters it should have same effect. 
attached diff.

current:
if (SDL_GameControllerHasRumble(game_controller_)) {  
change to :
if (0==SDL_GameControllerRumble(game_controller_, 0, 0, 0)) {

for backward compatibility please don't use SDL_GameControllerHasRumble.

attached diff

[game_controller.diff.txt](https://github.com/ros-drivers/joystick_drivers/files/12779907/game_controller.diff.txt)

meyerj, 2023-10-13:
Same problem for me when building ROS Humble from source in Ubuntu 20.04 Focal (a [Tier 3 platform](https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html#system-requirements)):

```
[ 50%] Building CXX object CMakeFiles/game_controller.dir/src/game_controller.cpp.o
/opt/ros/humble/src/joystick_drivers/joy/src/game_controller.cpp: In member function ‘void joy::GameController::handleControllerDeviceAdded(const SDL_ControllerDeviceEvent&)’:
/opt/ros/humble/src/joystick_drivers/joy/src/game_controller.cpp:319:7: error: ‘SDL_GameControllerHasRumble’ was not declared in this scope; did you mean ‘SDL_GameControllerRumble’?
  319 |   if (SDL_GameControllerHasRumble(game_controller_)) {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       SDL_GameControllerRumble
make[2]: *** [CMakeFiles/game_controller.dir/build.make:63: CMakeFiles/game_controller.dir/src/game_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:167: CMakeFiles/game_controller.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

Installed versions:
```
$ dpkg -l | grep libsdl2
ii  libsdl2-2.0-0:amd64                                 2.0.10+dfsg1-3                        amd64        Simple DirectMedia Layer
ii  libsdl2-dev:amd64                                   2.0.10+dfsg1-3                        amd64        Simple DirectMedia Layer development files
```
and `sdl2_vendor` selects the system version by default.

If this is the only detail, then I agree with @mor2code that the `joy` package should probably be kept backwards-compatible by either using the old API or by compiling conditionally (the version can be checked by including `SDL2/SDL_version.h`).

I don't see how https://github.com/ros-drivers/joystick_drivers/pull/270 would solve the problem (and it does not) as long as `SDL2_FOUND` is true and we use the system version. I tested that patch, without success.

The pull request that introduced the new API: https://github.com/ros-drivers/joystick_drivers/pull/258

This repository also has a [`humble-devel`](https://github.com/ros-drivers/joystick_drivers/tree/humble-devel) branch that is not affected, but the [rosdistro index](https://github.com/ros/rosdistro/blob/f6853594cef801828e29cc90c21c3ac6cdb954af/humble/distribution.yaml#L2632) and https://index.ros.org/p/joy/#humble point to the `ros2` branch for Humble?

clalancette, 2023-10-23:
Closing this as it was fixed by #273 .

> [](https://github.com/ros-drivers/joystick_drivers/pull/258)
>
> This repository also has a [`humble-devel`](https://github.com/ros-drivers/joystick_drivers/tree/humble-devel) branch that is not [...]

Yeah, we can probably just remove that `humble-devel` branch, as we aren't currently using it (as far as I know).  I'll take a quick look and then do that.

PR #270 (2023-10-12): Upgrade to SDL2.0.20
Fix #267
PR #273 (2023-10-23): Fix compatibility with SDL 2.0.10 for Ubuntu Focal
Resolves #267.

`SDL_GameControllerHasRumble()` was introduced in SDL 2.0.18 only (https://github.com/libsdl-org/SDL/pull/4943).
closed 2023-10-23 by clalancette