ros-drivers/joystick_drivers #274: joy 3.2.0-1 failing to build on Humble RHEL 8

nuclearsandwich, 2023-10-30:
Since the release of joystick_drivers 3.2.0-1 (ros/rosdistro#38750) the build of the `joy` package has been failing on RHEL with the following error:

The dependency seems to be properly declared and installed (`SDL2-devel-2.0.10-2.el8.x86_64` on RHEL 8) and has not recently changed so I suspect a usage issue between the previous release and this one.

```
01:10:59 DEBUG: /builddir/build/BUILD/ros-humble-joy-3.2.0/src/game_controller.cpp: In member function 'void joy::GameController::handleControllerDeviceAdded(const SDL_ControllerDeviceEvent&)':
01:11:02 DEBUG: /builddir/build/BUILD/ros-humble-joy-3.2.0/src/game_controller.cpp:319:7: error: 'SDL_GameControllerHasRumble' was not declared in this scope
01:11:02 DEBUG:    if (SDL_GameControllerHasRumble(game_controller_)) {
01:11:02 DEBUG:        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
01:11:02 DEBUG: /builddir/build/BUILD/ros-humble-joy-3.2.0/src/game_controller.cpp:319:7: note: suggested alternative: 'SDL_GameControllerRumble'
01:11:02 DEBUG:    if (SDL_GameControllerHasRumble(game_controller_)) {
01:11:02 DEBUG:        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
01:11:02 DEBUG:        SDL_GameControllerRumble
01:11:02 DEBUG: make[2]: *** [CMakeFiles/game_controller.dir/build.make:79: CMakeFiles/game_controller.dir/src/game_controller.cpp.o] Error 1
01:11:04 DEBUG: make[2]: Leaving directory '/builddir/build/BUILD/ros-humble-joy-3.2.0/.obj-x86_64-redhat-linux-gnu'
01:11:04 DEBUG: make[1]: Leaving directory '/builddir/build/BUILD/ros-humble-joy-3.2.0/.obj-x86_64-redhat-linux-gnu'
01:11:04 DEBUG: make[1]: *** [CMakeFiles/Makefile2:252: CMakeFiles/game_controller.dir/all] Error 2
01:11:04 DEBUG: make: *** [Makefile:149: all] Error 2
01:11:04 DEBUG: RPM build errors:
01:11:04 DEBUG: error: Bad exit status from /var/tmp/rpm-tmp.zYwS7B (%build)
01:11:04 DEBUG:     Bad exit status from /var/tmp/rpm-tmp.zYwS7B (%build)
01:11:04 DEBUG: Child return code was: 1
```

To reproduce the package build locally you can use [ros-infrastructure/ros_buildfarm](https://github.com/ros-infrastructure/ros_buildfarm).
```
mkdir /tmp/rhel-joy
generate_release_script.py https://raw.githubusercontent.com/ros2/ros_buildfarm_config/ros2/index.yaml humble rhel joy rhel 8 x86_6 > /tmp/rhel-joy/run.sh
cd /tmp/rhel-joy
sh run.sh
```

clalancette, 2023-10-31:
Yeah, we have a fix for this merged, I just need to release it.  I'll go ahead and do that.

clalancette, 2023-10-31:
And this is now building on https://build.ros2.org, so closing.

closed 2023-10-31 by clalancette