ros-drivers/joystick_drivers #278: Bug: Joystick Detection Issue with ROS2 Joy 3.3.0 in Docker

mariamercade, 2024-02-02:
I am using the ROS2 joy package within a Docker container, based on the ROS2 Humble image, while sharing the volume /dev/input. I've encountered a bug in the latest version 3.3.0 of joy. Essentially, if my joystick (which operates via Bluetooth) is not powered on before executing docker run, the joy node never detects it even after it's connected. I conducted a test by reverting back to version 3.1.0 by cloning the source repository, and this issue did not occur: I could execute docker run with the joystick powered off, then power it on, and it would be detected.

It seems like some of the recent changes is causing this behavior.

clalancette, 2024-02-02:
Can you please report this again over at https://github.com/ros-drivers/joy?  That is where the source repository for this code.  Thank you!

mariamercade, 2024-02-02:
The repo you said does not exist and in ROS index the URI to the joy pkg is this one.

clalancette, 2024-02-02:
> The repo you said does not exist and in ROS index the URI to the joy pkg is this one, @clalancette .

Sorry, it is https://github.com/ros-drivers/joystick_drivers .  Can you point me to where it says this is the right repository in rosindex?  That shouldn't be the case.

mariamercade, 2024-02-05:
This is the link to the ROS index: https://index.ros.org/p/joy/

The link you have provided is the same as the one specified in the ROS index and is the same as the one to which I have reported this issue. Can you reopen it, @clalancette ?

clalancette, 2024-02-06:
Sorry, you are right.  No matter how many times I looked at this, I always thought that this was in the wrong repository.  My mistake, I apologize.  I'll reopen.

JWhitleyWork, 2024-02-22:
@mariamercade In the past, I have always had to use `joy_linux` instead of `joy` in Docker containers. I am pretty sure this has to do with the low-level API being used by each node - SDL2 (which is mostly platform-agnostic) vs a Linux-specific joystick API. Please try this and report back.

clalancette, 2024-02-22:
That said, this does look like a regression on something that used to work.

My best guess is that this is related to #247 or #261 , which are the only ones that changed thing in `joy` (other than vendoring a newer version of SDL2).  @mariamercade Can you try reverting those PRs one at a time and see if it fixes things for you?

mariamercade, 2024-02-26:
Reverting #247 worked for me, @clalancette .

It also works with `joy_linux`, @JWhitleyWork.