ros-drivers/joystick_drivers #280: Unable to compile https://github.com/ros-drivers/joystick_drivers/tree/main on debian bullseye

firth650, 2024-02-25:
Hi
    I am using ROS2 Humble on an RPI4 Debian bullseye. I realise that this is not officially supported, but seems to work quite well otherwise. 
I'm trying to add "joy" from https://github.com/ros-drivers/joystick_drivers/tree/main, however I cannot get it to compile. I suspect that I have a missing dependency but I'm not sure what: The error messages are as below

--- stderr: spacenav_node                                                                               
Traceback (most recent call last):
  File "/home/robot/ros2_humble/joystick/joystick_drivers/build/spacenav_node/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('/home/robot/ros2_humble/joystick/joystick_drivers/build/spacenav_node/devel/env.sh')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/catkin/environment_cache.py", line 63, in generate_environment_script
    env_after = ast.literal_eval(output.decode('utf8'))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ast.py", line 64, in literal_eval
    node_or_string = parse(node_or_string.lstrip(" \t"), mode='eval')
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ast.py", line 50, in parse
    return compile(source, filename, mode, flags,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<unknown>", line 1
    ROS_DISTRO was set to 'humble' before. Please make sure that the environment does not mix paths from different distributions.
               ^^^
SyntaxError: invalid syntax
CMake Error at /usr/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python3
  "/home/robot/ros2_humble/joystick/joystick_drivers/build/spacenav_node/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /usr/share/catkin/cmake/all.cmake:221 (safe_execute_process)
  /usr/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:6 (find_package)

---
Failed   <<< spacenav_node [3.19s, exited with code 1]
Aborted  <<< joy [3.28s]
Aborted  <<< wiimote [3.11s]

Summary: 0 packages finished [3.91s]
  1 package failed: spacenav_node
  2 packages aborted: joy wiimote
  3 packages had stderr output: joy spacenav_node wiimote
  1 package not processed

closed 2024-02-27 by firth650

BerenChang, 2024-12-04:
I run into the same problem. How did you solve it?

firth650, 2024-12-04:
I did not try to fix this. I only wanted to use a ps4 controller not a wiimote, so I simply removed the wiimote package dependency.