ros-drivers/joystick_drivers #297: If starting joy before xbox controller - no controller is detected

drozzy, 2025-02-11:
If I start my ros2 joy node before I turn on the bluetooth xbox controller then joy doesn't detect it.

Steps to reproduce:

1. Start joy node: `ros2 launch joy teleop_joy_node`
2. Turn on xbox controller
3. Start `ros2 topic echo joy`
4. No output

However, this works just fine:

1. Turn on xbox controller
2. Start joy node: `ros2 launch joy teleop_joy_node`
3. Start `ros2 topic echo joy`
4. Output present.

Consequently, if the controller is ever turned-off during operation, we are back to the first problem - it's not detected upon starting.

I would love if there was some way to force joy node to somehow keep reconnecting to it.

drozzy, 2025-07-21:
Nope.

jlack1987, 2025-09-22:
Also having this issue